fossasia/pslab-desktop #710: Archive this repository in favour of https://github.com/fossasia/pslab-python

froscon2024, 2025-03-04:
This software project is not developed for a long time and it looks to be dead. The problem is that pslab users are still pointed to this repository here and not to https://github.com/fossasia/pslab-python
Github have the capability to archive a repo. It would be also great to add to the readme a huge message that the people should go to https://github.com/fossasia/pslab-python if they want a desktop software for their computer instead of having to use the smartphone.

bessman, 2025-03-04:
Makes sense. We would like to provide a graphical front-end on the desktop, but the fact is we do not currently have the resources to do so. If and when we get around to it, this repo could be unarchived.

@mariobehling what do you think?

froscon2024, 2025-03-04:
> this repo could be unarchived

Often when source code is not touched for that long time, it make most sense to restart from zero. 
A software with gui would make sense to be build with using the pslab-python code as base and just be a gui for the pslab-python. Then you wont have to do parts that do the same thing twice and developing the gui would also push the development of the pslab-python base.

bessman, 2025-03-04:
> A software with gui would make sense to be build with using the pslab-python code as base and just be a gui for the pslab-python.

As a point of historical interest, pslab-python used to include a GUI a long time ago. That GUI was replaced by the one in the current repository.

Whether we unarchive or replace this repo would depend on whether the hypothetical new GUI is a continuation of this one or a new one entirely.